adafruit/Adafruit_CircuitPython_SSD1306 #1: Does not work with 0.9.1 due to missing framebuf module

shenki, 2017-03-21:
```
>>> uos.uname()
(sysname='samd21', nodename='samd21', release='0.9.1', version='0.9.1 on 2017-02-28', machine='Adafruit Feather M0 Basic with samd21g18')
>>> import adafruit_ssd1306 as ssd1306
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/flash/lib/adafruit_ssd1306/__init__.py", line 1, in <module>
  File "adafruit_ssd1306/ssd1306.py", line 3, in <module>
ImportError: no module named 'framebuf'
```

According to the release notes, framebuf was removed from CircuitPython in 0.8.1:

> * Remove the `framebuf`. It made the binaries too large with all of the extra functionality added in 1.8.7.

https://blog.adafruit.com/2017/01/12/circuitpython-0-8-1-released/

shenki, 2017-03-21:
I assume this module was written to fix the issue:

 https://github.com/adafruit/micropython-adafruit-framebuf/releases

The documentation at https://learn.adafruit.com/micropython-hardware-ssd1306-oled-display?view=all needs updating

tannewt, 2017-03-24:
I've updated @tdicola's guide here: https://learn.adafruit.com/micropython-hardware-ssd1306-oled-display/software#adafruit-circuitpython-module-install

Thanks for the issue!

closed 2017-03-24 by tannewt

owendelong, 2018-03-25:
@tannewt Um the URL where you claim to have updated @tdlcola's guide is a 404.

ladyada, 2018-03-26:
try https://learn.adafruit.com/micropython-hardware-ssd1306-oled-display/