adafruit/Adafruit_CircuitPython_SSD1306 #7: not working with 128x64 product 938

dhalbert, 2018-01-05:
A user can't get much working with this 128x64 OLED: https://www.adafruit.com/product/938.
See https://forums.adafruit.com/viewtopic.php?f=60&t=128946.

I happen to have one of these and have confirmed the problem. `.invert()` works, and so does `.poweron()` and `.poweroff()`, but nothing else really. Reconfirmed that library works fine with a 128x32 FeatherWing.

deshipu, 2018-01-05:
Are we sure it's SSD1306 and not SH1106? The latter has very similar commands (including invert and power), but different way of accessing the display memory (SH1106 uses pages). I had exactly the same problem myself, and it turned out the display is SH1106 and not SSD1306.

I have a driver for the SH1106 for MicroPython that you could try and see if it works: https://bitbucket.org/thesheep/micropython-oled/src/3caad24615e16343723ccd46eddb3a615be502a6/sh1106_i2c.py?at=default&fileviewer=file-view-default

dhalbert, 2018-01-05:
I'm adding reset-pin support to the library and it looks like it's all working out. The FeatherWing OLED has hardware reset support, so testing didn't encounter this issue. I just have to re-test the other combinations to make sure I didn't break something else.

deshipu, 2018-01-05:
Are you saying that it works when you reset it through the hardware pin?

dhalbert, 2018-01-05:
Yep, doing a reset makes it work fine. Part of `__init__()` is to call `.poweron()`. In the SPI subclass this toggles the reset pin. In the I2C class it just sends an I2C command. I made passing the reset possible (and optional) to the I2C instance, and if it exists, it will toggle it as well. I'll have a PR soon.

deshipu, 2018-01-05:
I wonder if we are missing something in the initialization sequence then — software reset should work as well.

ladyada, 2018-01-05:
nah, its def a hardware thing. basically these OLEDs need a clean reset on power on, for the featherwing i add a little resetter chip, but low cost or earlier-designed OLEDs didnt have it, so you have to manually reset.

PR #8 (2018-01-05): Provide reset pin support for I2C
Fixes #7. Tested on an Adafruit 938 128x64 display, both with I2C and SPI. Also tested with a FeatherWing OLED with I2C.
closed 2018-01-05 by ladyada