adafruit/Adafruit_CircuitPython_SSD1306 #12: "No module named 'framebuf'" error running simpletest.py on RaspberryPi

ngmacha, 2018-11-16:
I'm trying to use the SSD1306 display on RaspberryPi (latest Stretch image) with CircuitPython. 
Installed Blinka and other libraries as following:
`pip3 install RPI-GPIO`
`pip3 install adafruit-blinka`
`pip3 install Adafruit_CircuitPython_SSD1306`

The Blinka test was OK, however running the simpletest.py gives the error that framebuf module is not found. Tried to install additional dependencies, as per documentation (https://cdn-learn.adafruit.com/downloads/pdf/micropython-hardware-ssd1306-oled-display.pdf?timestamp=1542370878):
`pip3 install adafruit-circuitpython-busdevice`
`pip3 install adafruit-circuitpython-register`

Tried to follow the issue #1, but very confusing to understand how to install the framebuf.
Please help. Thanks - Mac Ha

jerryneedell, 2018-11-16:
Try this:
get bitmapfont.py and font5x8.bin from https://github.com/adafruit/micropython-adafruit-bitmap-font  and put them in the same folder as this test code - saved as oled_test.py
```
# Import pins.
from board import SCL,SDA
from busio import I2C
import time
# Import the SSD1306 module.
import adafruit_ssd1306
import bitmapfont
i2c=I2C(SCL,SDA)
display = adafruit_ssd1306.SSD1306_I2C(128, 32, i2c)
display.fill(0)
display.show()
bf = bitmapfont.BitmapFont(128, 32, display.pixel)
bf.init()
bf.text('Hello, World!', 0, 0, 1 )
bf.text('Hello, World!', 0, 10, 1)
bf.text('Hello, World!', 10, 20, 1)
display.show()
time.sleep(2)
display.invert(True)
time.sleep(2)
display.invert(False)
time.sleep(2)
display.fill(0)
display.show()
```

I am able to then run this vis:
python3  oled_test.py

jerryneedell, 2018-11-16:
sorry -- also put framebuf.py  from https://github.com/adafruit/micropython-adafruit-framebuf in the same folder as the example and bitmapfont.py

ladyada, 2018-11-16:
just FYI because of all the complexity involved we're not quite read to support circuitpython for oled displays on linux. it *might* work :) but its not tested at all! we have a plain Python repo for now, with example code. long term, yes we'll be moving to this library only

jerryneedell, 2018-11-16:
Fair enough --  but it does work ;-) 
[image]

ladyada, 2018-11-16:
rad :)

ladyada, 2018-11-16:
jerryn, is that with framebuf? ar eyou using PIL to draw or just main framebuf - thats the thing we'd have to document is how to use PILlow

jerryneedell, 2018-11-16:
That is with framebuf.py

ngmacha, 2018-11-20:
Sorry, looks like the display itself is faulty (fake, I guess), as only the same 32 lines shows up with own DigiSpark's OLED demo sketch... So I've already ordered a couple of Adafruit's 1.3 inch 128x64 OLEDs.
I think this issue can be now closed?