adafruit/Adafruit_CircuitPython_SSD1306 #15: not working wth OLED Featherwing on nrf52840

jerryneedell, 2018-12-22:
I just tried running the test  cases on a particle argon (nrf52840) with an OLED featherwing and I just get a screen full of random pixels. The same code works fine on a metro_m4 _express

All of the test cases (bouncing bali,simpletest and ssd1306_framebuftest have the same results so it is not just framebuf)
Note -- the default nrf builds still include the "built in" framebuf  but I tried it with they built-in and the new adafruit_framebuf.py -- same results.

I have not test other nrf52840 boards yet.

jerryneedell, 2018-12-22:
It works fine with the SPI interface to a 128x32 SPI OLED -- just not working with I2C to the featherwing -- SPI does have a RESET so that may be the issue -- is there a way to get a RESET signal to the OLED featherwing?

deshipu, 2018-12-22:
Can you verify that it works after you press the reset button on the wing manually?

jerryneedell, 2018-12-22:
It does not. That also resets the argon.

jerryneedell, 2018-12-22:
I will try later with the boards separated.

jerryneedell, 2018-12-23:
separating the boards and pressing RESET or setting the RESET pin to a GPIO pin and passing it to the configuration makes no difference - still just a lot of white pixels.

jerryneedell, 2018-12-23:
I verified the same behavior on a PCA10056.

ladyada, 2018-12-28:
closing this in favor of the https://github.com/adafruit/circuitpython/issues/1422

closed 2018-12-28 by ladyada