adafruit/Adafruit_CircuitPython_SSD1306 #23: TypeError: function missing 1 required positional arguments

tannewt, 2019-01-10:
https://forums.adafruit.com/viewtopic.php?f=60&t=145925

ladyada, 2019-01-10:
weird, its definitely a something that happened when i moved to adafruit_framebuf because if i remove the 'import framebuf' and just use adafruit_ its ok, but the error isnt something i can see...like the error line number isnt where the error is?

jerryneedell, 2019-01-10:
I think this error occurs if the user has the old framebuf.py file on their board.

ladyada, 2019-01-10:
in 3.x its frozen in... 😐

jerryneedell, 2019-01-10:
yes - unless they copied it over for some reason....

dhalbert, 2019-01-10:
(If this is the issue... not sure): If you move it out of lib and into the top level of CIRCUITPY it will get imported before the frozen one without messing with the standard`sys.path`.
```
>>> sys.path
['', '/', '.frozen', '/lib']
>>> 
```

caternuson, 2019-01-10:
Just confirming it's something in the SSD1306 lib. And not a frozen vs. local thing.

I recreated this. Feather M0 + OLED FeatherWing, all clean and latest everything:
```python
Adafruit CircuitPython 3.1.2 on 2019-01-07; Adafruit Feather M0 Express with samd21g18
>>> from board import SCL, SDA
>>> import busio
>>> import adafruit_ssd1306
>>> i2c = busio.I2C(SCL, SDA)
>>> display = adafruit_ssd1306.SSD1306_I2C(128, 32, i2c)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function missing 1 required positional arguments
>>> adafruit_ssd1306.__version__
'2.6.2'
```
I then stepped through previous versions of `adafruit_ssd1306` and got it to work at the 2.5.0 release.
```python
Adafruit CircuitPython 3.1.2 on 2019-01-07; Adafruit Feather M0 Express with samd21g18
>>> from board import SCL, SDA
>>> import busio
>>> import adafruit_ssd1306
>>> i2c = busio.I2C(SCL, SDA)
>>> display = adafruit_ssd1306.SSD1306_I2C(128, 32, i2c)
>>> adafruit_ssd1306.__version__
'2.5.0'
```

ladyada, 2019-01-10:
nah its def using the built in one! thing is im not sure what line is actually failing :/

caternuson, 2019-01-10:
I think it's CP doing something with the inheritance and the `__init__` parameters. It's failing right at the instantiation as it's expecting...??? dunno, but not per code. A different failure:
```python
Adafruit CircuitPython 3.1.2 on 2019-01-07; Adafruit Feather M0 Express with samd21g18
>>> from board import SCL, SDA
>>> import busio
>>> import adafruit_ssd1306
>>> i2c = busio.I2C(SCL, SDA)
>>> display = adafruit_ssd1306.SSD1306_I2C(128, 32, i2c, addr=0x3d)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function does not take keyword arguments
>>> 
```
FWIW - It works on RPi.
```python
pi@raspberrypi:~ $ python3
Python 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from board import SCL, SDA
>>> import busio
>>> import adafruit_ssd1306
>>> i2c = busio.I2C(SCL, SDA)
>>> display = adafruit_ssd1306.SSD1306_I2C(128, 64, i2c, addr=0x3d)
>>> display.fill(1)
>>> display.show()
>>> 
```

ladyada, 2019-01-10:
yeah thats cause on pi its using `adafruit_framebuf` - i messd with this for 30 minutes and couldn't figure it out. its something with the framebuf instantiation but according to the docs im doing it right :/

eudoxos, 2019-01-26:
Is there any update on this? I am facing the same with the latest circuitpython 3.1.2 (cricket varianant, on Feather M0 Express) and latest adafruit_ssd1306 2.6.2. If I pass e.g. a reset pin as the 4th positional argument, I get `TypeError: object with buffer protocol required`. Can seomeone from Adafruit care to look at this, or at least state which exact "older version" of the module will work? (suggested in the forum post)

ladyada, 2019-01-26:
we dont know but we're also moving away from framebuf completely - so its a bit moot and probably not worth us doing a lot of hacking on this. next generation display stuff will be displayio only
note you wont have enough memory on a feather m0 to run crickit and an oled. a feather m4 will do fine and then you can install `adafruit_framebuf` on cpy 4.x and this all goes away

eudoxos, 2019-01-26:
Anyway, I would just appreciate documented workaround till cpy 4.0 is released.

I use crickit purely for memory considerations, thinking it will save some memory as I need to use use oled+motor. Anyway, M4 is on the way already and I will throw both M0s away happily.

eudoxos, 2019-01-26:
PS as someone who is not buying cheap clones to support your development, I'd expect better support than "your issue is a bit moot and it is not worth for us to spend time on it".

ladyada, 2019-01-26:
@eudoxos hiya we responded to your post in under an hour and we have a workaround for it, use `adafruit_framebuf` with circuitpython 4.x

eudoxos, 2019-01-26:
I did not understand cpy 4.0 can be installed already. I will try, thanks.

ladyada, 2019-01-28:
try deleting any 'framebuf.py' or 'framebuf.mpy' file

Moving-Electrons, 2019-01-28:
@ladyada I used a CP 4.0 build from 2019-01-27 (provided by Jerryn) and it worked fine (the 2019-01-23 build didn't work, though). Also, to make it work, I had to stop using the neopixel library. Apparently the M0 doesn't have enough memory to handle everything.

tannewt, 2019-01-29:
@eudoxos 4.0.0 Beta 1 is available here: https://github.com/adafruit/circuitpython/releases It's pretty stable despite being marked unstable.

kenidowa, 2019-02-20:
I was seeing this same issue on 3.1.2 when I was doing a mpy-cross of the adafruit-ssd1306.py file but not when I used the original .mpy for 3.0 lib.  I am playing around with a sh1106 oled and was modifying the ssd1306.py now since this sounds fixed in 4.0 beta-2 is there a mpy-cross for 4.0 available to try with the sh1106.py file I am trying to use?  I have downloaded 4.0beta2 and the associated lib.  I am using a raspbian-stretch version.

ladyada, 2019-02-20:
this thread isn't relevant for raspbian - on linux you will always end up using adafruit_framebuff. we don't have an sh1106 library tho. if you write one please submit it to the bundle!

kenidowa, 2019-02-20:
Let me clarify comment.  I am using the OLED display on a itsy-bitsy M4 and on a Trinket M0.  The SH1106 code I am trying to use is from your Contributed Bundles.  I am using the Raspberry Pi as my host computer and was included the Raspbian-Stretch as a reference for the MPY-CROSS to convert the SH1106.py code to a mpy version to add to the lib file on my Itsy-Bitsy M4 using the 4.0 Beta 2 version of CircuitPython.  Currently using the adafruit_ssd1306.mpy I get a bar about an 8th of the display high when I do a fill.  I hope this clarifies my intent.  Thank for you quick response. Ken

ladyada, 2019-02-20:
i'd say try to get it working on the M4 first, with 4.x beta 2 - you dont need to mpy the library on an M4 as there's plenty of memory

kenidowa, 2019-02-20:
Thanks for the input.  I made those changes to the Contributed Bundles file for sh1106 and it WORKS!  Loaded in lib on board as adafruit_sh1106,py.  Need to try more things but bouncing_ball.py works from the contibuted examples for it.  Running on an ItsyBitsyM4.

ladyada, 2019-02-20:
awesome! when you have a chance please submit a PR or issue fix to that library so others can use it too :)

kenidowa, 2019-02-20:
Doing some testing.  Looks like some of the functions such as rect, text from adafruit_framebuf don't work yet.  Will look into issue fix after more testing.  PR?

ladyada, 2019-02-20:
they should ? but yeah send a PR if you find any bugs :)

kenidowa, 2019-03-01:
I have found 2 issues with this but I don't think they are in the sh1106,py code.
rect in framebuf causes an Attribute Error:  SH1106_I2C object has no attribute 'rect'.
scroll in framebuf causes a TypeError: function takes 1 positional arguments but 3 were given.
I don't have a SD1306 display so haven't try these on one.  They may have the same problems.  Looking at adafruit_framebuf I don't see what would cause these errors.  All other functions I have tried work including text.

inferno4you, 2019-04-28:
My guess is when I add None to the 4 positional argument it is looking for a `TypeError: object with buffer protocol required`.

peppekerstens, 2019-06-10:
FWIW:

**Hardware**
- ItsyBitsy M4 Express
- Lolin v1.1.0 OLED shield

**Software**
- CircuitPython 3.1.2

**Issue**
the same as described. 

**Fix**
Basically: https://forums.adafruit.com/viewtopic.php?f=60&t=145925#p726277

- latest CircuitPython (4.0.1 on 2019-05-22)
- https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/download/20190608/adafruit-circuitpython-bundle-4.x-mpy-20190608.zip

Code used
from: https://github.com/adafruit/Adafruit_CircuitPython_SSD1306/tree/master/examples
(https://github.com/adafruit/Adafruit_CircuitPython_SSD1306/commit/1f6b41f4b5f44e05f4c4602b6f7edadb0dfc4791)

Needs
- adafruit_bus_device
- adafruit_framebuf.mpy
- adafruit_ssd1306

**Remaining**
- Examples are not complete/need links/instructions to used PIL library in some examples for the complete NOOB (me)
- No (clear) examples on how to use/produce text

I vote this issue to be closed as this a valid and working for the issue itself (at least on latest CPython build(s))

kattni, 2019-06-11:
@peppekerstens Thank you for all the details! I will close this issue.

closed 2019-06-11 by kattni