adafruit/Adafruit_CircuitPython_SSD1306 #24: SSD1306_SPI TypeError: function expected at most 5 arguments, got 6

everett1992, 2019-02-16:
I'm following [this spi oled guide](https://learn.adafruit.com/adafruit-oled-featherwing/circuitpython-and-python-usage#spi-initialization-8-15) to program my [128x32 SPI oled](https://learn.adafruit.com/2-3-monochrome-128x32-oled-display-module/overview) from a [feather m0 express](https://www.adafruit.com/product/3403).

I am running circuit python [3.1.2](https://github.com/adafruit/circuitpython/releases/tag/3.1.2), with the [3.x bundle](https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/tag/20190216)

```
# boot_out.txt
Adafruit CircuitPython 3.1.2 on 2019-01-07; Adafruit Feather M0 Express with samd21g18                          
```

I'm running the sample code, with the pins changed to mach how I've connected the wires.
```
# code.py
import adafruit_ssd1306
import board
import busio
import digitalio
 
spi = busio.SPI(board.SCK, MOSI=board.MOSI)
dc_pin = digitalio.DigitalInOut(board.A0)    # any pin!
reset_pin = digitalio.DigitalInOut(board.A2) # any pin!
cs_pin = digitalio.DigitalInOut(board.A1)    # any pin!
 
oled = adafruit_ssd1306.SSD1306_SPI(128, 32, spi, dc_pin, reset_pin, cs_pin)
```

I cannot initialize SSD1306_SPI, it fails with `TypeError: function expected at most 5 arguments, got 6`

```
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 11, in <module>
TypeError: function expected at most 5 arguments, got 6
```

Before reporting this bug I updated circuitpython to 3.1.2 by copying the .uf2 into the bootloader, and I cleared the filesystem using `system.erase_filesystem()`, then I reinstalled the circuit python bundle.

ladyada, 2019-02-16:
odd - it should work, not sure what it is :/ can you use i2c?

everett1992, 2019-02-16:
Will you update the guides to say that spi doesn't work?

jerryneedell, 2019-02-16:
@everett1992 just curious -- when did you last update the "bundle"? If it has been awhile, try updating to the current 3.x release.

everett1992, 2019-02-16:
@jerryneedell Today, extracted from `adafruit-circuitpython-bundle-3.x-mpy-20190216.zip`, downloaded from https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/tag/20190216,
linked from https://learn.adafruit.com/adafruit-feather-m0-express-designed-for-circuit-python-circuitpython/circuitpython-libraries

jerryneedell, 2019-02-16:
FYI - I just tried the code above on a Feather_m4_express with CP 4.0 beta  and it works normally
```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-beta.1-7-g765d877df on 2019-01-28; Adafruit Feather M4 Express with samd51j19
>>> import adafruit_ssd1306
>>> import board
>>> import busio
>>> import digitalio
>>> spi = busio.SPI(board.SCK, MOSI=board.MOSI)
>>> dc_pin = digitalio.DigitalInOut(board.A0)    # any pin!
>>> reset_pin = digitalio.DigitalInOut(board.A2) # any pin!
>>> cs_pin = digitalio.DigitalInOut(board.A1)    # any pin!
>>> oled = adafruit_ssd1306.SSD1306_SPI(128, 32, spi, dc_pin, reset_pin, cs_pin)
>>> oled.invert(True)
>>> oled.invert(False)
>>> oled.invert(False)

```
I realize that may not help you, but the 3.x and 4.x libraries "should" be the same.
Not sure what is wrong.

jerryneedell, 2019-02-16:
one suggestion is to make sure you do not have an "old" version of adafruit_ssd1306.py or .mpy on your feather_m0. I've gotten caught by that.

kyp44, 2019-02-16:
I am having the same issue. It seems to be something odd happening where the SPI object is trying to use it's base class's constructors. I am running CP 3.1.2 with the 3.x bundle as well.

kyp44, 2019-02-16:
I am definitely not using an old version of adafruit_ssd1306.py. I am using the latest .py version from GitHub directly (not using the .mpy) and if I remove it I get a module not found error.

everett1992, 2019-02-16:
@jerryneedell I don't have an old version of adafruit_ssd1306.py, I cleared the filesystem with `storage.erase_filesystem()` before filing the ticket.

I've installed cp 4.x and it's 4.x libraries and now the code is working.

Is is possible that the feb 16th 2019 release of cp 3.x is bad?
https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/tag/20190216

kyp44, 2019-02-16:
Like everett1992 I am using a Feather M0 express board so maybe it's something specific to this board.

jerryneedell, 2019-02-16:
hmm-- glad it is working under 4.x  -- clearly some issue with either 3.1.2 and/or the 3.x library.  That's a great clue.

kyp44, 2019-02-16:
everett1992, what 4.0.0 beta version are you using? With Beta 2 the framebuf has been removed so it cannot find that module.

everett1992, 2019-02-16:
@kyp44 4.0.0-beta.2 

[adafruit-circuitpython-feather_m0_express-en_US-4.0.0-beta.1.uf2](https://github.com/adafruit/circuitpython/releases/download/4.0.0-beta.1/adafruit-circuitpython-feather_m0_express-en_US-4.0.0-beta.1.uf2) from the [circuitpython/releases page](https://github.com/adafruit/circuitpython/releases)

everett1992, 2019-02-16:
The display is working but the image is poor. Calling `oled.fill(1) oled.show()` only draws every other line, with a few dead pixels on the end. But this feels like a hardware issue, not a software issue.

ladyada, 2019-02-17:
ever, sounds like its a mixup with chipsets or resolutions!

kyp44, 2019-02-17:
@everett1992 I'm seeing the exact same thing on my screen. I'm using the 128x32 OLED. You?

kyp44, 2019-02-17:
[@everett1992 FYI I figured out the screen issue (with my screen anyway). Mine uses the SSD1305 chip, which supports screen resolutions of up to 132x64. Evidently the way my screen is setup of that the chip is expecting a buffer of this size even though the actual screen is smaller (128 x 32).

It worked wonderfully when I set the resolution to 132 x 64 in the SSD1306_SPI construtor. Then pixel 0,0 on the actual screen (upper left is) actually 4,32 in the frame buffer. So setting turning that pixel on will set the upper-left-most pixel on the actual screen. The lower-right pixel is then 131,63 as you'd expect.

everett1992, 2019-02-17:
@kyp44 Thanks that's working great now.

@ladyada Should the docs mention the framebuffer size?

I'm fine closing this issue, but someone should look into the issues with this library in circuit python 3.1.2.

makermelissa, 2019-09-26:
Closing because CPy 3.1.2 is not supported anymore. If this issues occurs in more recent versions, please feel free to reopen.

closed 2019-09-26 by makermelissa