adafruit/Adafruit_CircuitPython_SSD1306 #27: Examples: no clear instructions on where to get depended PIL library and/or no separate example  on text display

peppekerstens, 2019-06-10:
issue text should be clear enough (i hope) :)

ladyada, 2019-06-10:
@kattni please add a note to the top of the linux examples that they are for linux!

peppekerstens, 2019-06-11:
@ladyada and @kattni ; I am sorry, should have RTFM better;  after re-reading, the examples clearly say that they are for Linux.... :|

Would still be nice to include a example on how to accomplish text on 'pure' CircuitPython devices.  :)

ladyada, 2019-06-11:
https://github.com/adafruit/Adafruit_CircuitPython_SSD1306/blob/master/examples/ssd1306_framebuftest.py

https://learn.adafruit.com/adafruit-oled-featherwing/circuitpython-and-python-usage#text-8-27

kattni, 2019-06-11:
I took a look at it and found they had a note already as well. Glad we figured it out!

Closing.

closed 2019-06-11 by kattni