adafruit/Adafruit_CircuitPython_SSD1306 #50: Example code not helpful - draws black on black

herrmanthegerman, 2020-08-25:
I assume that the example code given on _README.rst_ should be considered a "smoke test" or "hello world" program to enable the user to check that his/her display can be addressed and is basically working.

That said, with executing `display.fill(0)` it draws a black background on my black display. I didn't see any change at all. 

I suggest to change this to `display.fill(1)` so it draws some other "color" than black on the display. At least this works for my SSD1306 OLED display.

Does that make sense?

ladyada, 2020-08-25:
sure, please submit a PR to revise it!

Wmyers559, 2020-09-11:
This should be fixed now, by mirroring the code in `examples/ssd1306_simpletest.py`.

closed 2021-04-13 by ladyada