adafruit/Adafruit_CircuitPython_SSD1306 #55: Not building per instructions

halxinate, 2021-01-17:
Tried to follow the new instructions here https://learn.adafruit.com/monochrome-oled-breakouts/python-setup

```
> pip3 install adafruit-circuitpython-ssd1306`
```
Getting in the end:
```
Downloading https://files.pythonhosted.org/packages/0c/d7/5d2f861155e9749f981e6c58f2a482d3ab458bf8c35ae24d4b4d5899ebf9/sysv_ipc-1.1.0.tar.gz (99kB)
    100% |████████████████████████████████| 102kB 298kB/s
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-lfr36pam/sysv-ipc/setup.py", line 11, in <module>
        import prober
      File "/tmp/pip-build-lfr36pam/sysv-ipc/prober.py", line 137
        d["SYSV_IPC_VERSION"] = f'"{version}"'
                                             ^
    SyntaxError: invalid syntax

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-lfr36pam/sysv-ipc/
```

halxinate, 2021-01-17:
N/M I guess. Followed your readme and it need s Python >3.5.0 
Just sad then.

ladyada, 2021-01-17:
https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi has instructions :)

closed 2021-01-17 by ladyada