adafruit/Adafruit_CircuitPython_SSD1306 #62: Suspected wrong value for Page Addressing Mode command.

snoring-pigeon, 2021-05-03:
Section 10.1.3 of the SSD1306 datasheet says that

"Set Memory Addressing Mode (20h)"

takes value

Page addressing mode (A[1:0]=10xb)"

but the SSD1306 driver uses 0x10 for Page Addressing. I think it should be 0x02.

https://github.com/adafruit/Adafruit_CircuitPython_SSD1306/blob/14739573f6a328d037f7693d20147ff0e943a057/adafruit_ssd1306.py#L102

jposada202020, 2021-05-03:
Thanks @snoring-pigeon I think it will depend on the display, I saw that configuration in some of the spreadsheets, however I have just tested this in the display and these are the results.
[image]

snoring-pigeon, 2021-05-04:
Thanks for checking it out; assuming I'm looking at the correct datasheet, the SSD1306 doesn't care about bits 7-2, so 0x10 should have the same effect as 0x00 (horizontal mode).

I'm still trying to get my head around the addressing; I never imagined there would be so many ways to organize so few bytes of RAM!

I'll close this, and reopen if needed when I understand more.

closed 2021-05-04 by snoring-pigeon