adafruit/Adafruit_CircuitPython_SSD1306 #70: Usage Example code requires two runs on Pico using Thonny before screen actually updates

roryjamesallen, 2021-11-07:
When running the example code (or seemingly any code using the module) on a Raspberry Pi Pico using Thonny IDE the code must be run twice before any changes to the display made in code are visible on screen.
It consistently is exactly 2 runs before changes are visible, and the changes are always visible after 2.

I have tried implementing a delay both before the refresh, and before a second code refresh, and neither fix the issue.

[image]

tannewt, 2021-11-08:
Does this occur when you use Mu or another editor? Thonny loads code in a unique way that may be causing this.

roryjamesallen, 2021-11-08:
That fixed it, thank you!

closed 2021-11-08 by roryjamesallen