adafruit/Adafruit_CircuitPython_SSD1306 #71: RPi Pico SSD 1306 I2C Serial Console Appears on Display

ditricsd, 2022-01-16:
Hi!

I'm trying to use an SSD1306 OLED display with RPi Pico using CircuitPython. My Problem is when I power on the RPi the test drawing works fine on the display, but after changing something in the code and saving to the RPi, the serial monitor appears on the OLED display (also works in the editor) and stops with an error, saying that the GP11 IO is in use (SCL connection for the display). The display only works as expected when I disconnect and reconnect the RPi:

Here is the init code:

   
    i2c = busio.I2C (scl=board.GP11, sda=board.GP10)
    if(i2c.try_lock()):
        print("i2c.scan(): " + str(i2c.scan()))
        i2c.unlock()
    print()   
    display_bus = displayio.I2CDisplay (i2c, device_address = 0x3C) # The address of my Board

    display = adafruit_displayio_ssd1306.SSD1306(display_bus, width=128, height=64)
    splash = displayio.Group()
    display.show(splash)

    color_bitmap = displayio.Bitmap(128, 64, 1) # Full screen white
    color_palette = displayio.Palette(1)
    color_palette[0] = 0xFFFFFF  # White
    
    bg_sprite = displayio.TileGrid(color_bitmap, pixel_shader=color_palette, x=0, y=0)
    splash.append(bg_sprite)

    color_bitmap = displayio.Bitmap(128, 64, 1) # Full screen white
    color_palette = displayio.Palette(1)
    color_palette[0] = 0xFFFFFF  # White
    
    bg_sprite = displayio.TileGrid(color_bitmap, pixel_shader=color_palette, x=0, y=0)
    splash.append(bg_sprite)
    
    # Draw a smaller inner rectangle
    inner_bitmap = displayio.Bitmap(118, 54, 1)
    inner_palette = displayio.Palette(1)
    inner_palette[0] = 0x000000  # Black
    inner_sprite = displayio.TileGrid(inner_bitmap, pixel_shader=inner_palette, x=5, y=4)
    splash.append(inner_sprite)
    
    # Draw a label
    text = "Nicolau dos"
    text_area = label.Label(terminalio.FONT, text=text, color=0xFFFF00, x=28, y=15)
    splash.append(text_area)

    text = "Brinquedos"
    text_area = label.Label(terminalio.FONT, text=text, color=0xFFFF00, x=32, y=25)
    splash.append(text_area)

Here is a video:
https://user-images.githubusercontent.com/80568222/149657931-ccd8c6b0-8c10-47a8-8b77-d96eee70a8b5.mp4

ladyada, 2022-01-16:
try add `displayio.release_displays()` at the first line of code?

ditricsd, 2022-01-16:
Cool! It works perfectly! Thank You!

closed 2022-01-16 by ditricsd