adafruit/Adafruit_CircuitPython_SSD1306 #72: Black screen with SPI

muranyia, 2022-04-06:
I can initialize the display but the example code gives me black screen. The same display works with I2C.
```
Adafruit CircuitPython 7.3.0-alpha.0-38-g862210b3f-dirty on 2022-03-24; Adafruit Feather M4 Express with samd51j19
Board ID:feather_m4_express
```
``` python
import board
import busio
import displayio
import adafruit_displayio_ssd1306
import sys

displayio.release_displays()

try: 
	spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)

	display_bus = displayio.FourWire(spi_bus=spi, command=board.D10, chip_select=board.D11, reset=board.D9, baudrate=10000000)

	display = adafruit_displayio_ssd1306.SSD1306(display_bus, width=128, height=64)

	splash = displayio.Group()
	display.show(splash)

	color_bitmap = displayio.Bitmap(120, 50, 1)
	color_palette = displayio.Palette(1)
	color_palette[0] = 0xFFFFFF # White

	bg_sprite = displayio.TileGrid(color_bitmap, pixel_shader=color_palette, x=0, y=0)
	splash.append(bg_sprite)

	while True:
		pass

except (KeyboardInterrupt) as e:
	spi.unlock()
	sys.exit(0)
```

ladyada, 2022-04-06:
is the display in SPI mode?

muranyia, 2022-04-06:
Yes, the jumper is on (contrary to the wrong label).

caternuson, 2022-04-19:
> Yes, the jumper is on (contrary to the wrong label).

By "on", do you mean the J1/J2 jumper traces are cut?
https://learn.adafruit.com/monochrome-oled-breakouts/wiring-128x64-oleds#step-3050993

muranyia, 2022-04-19:
A have soldered an actual jumper on the terminals. When I open the jumper, the display works fine with i2c. For SPI I close the jumper.

caternuson, 2022-04-19:
Were the jumper traces also cut?

makermelissa, 2022-04-19:
Hi, please contact the manufacturer for support with that display. Unfortunately we don't have the schematics or hardware so we're unable to debug. If you do figure it out and want to submit a PR, we'd be happy to take a look. Thanks.

closed 2022-04-19 by makermelissa

muranyia, 2022-04-19:
It was me soldering the jumpers together - your questions helped me find the problem. Thanks!

makermelissa, 2022-04-19:
Awesome. I'm glad it was something simple for you.

muranyia, 2022-10-11:
Yes. Jumper on, contrary to the wrong label.