adafruit/Adafruit_CircuitPython_SSD1306 #76: Initialize display cmd

TimonPeng, 2022-09-07:
Why initialize display cmd is big difference from with the previous repo and u8g2 too. Where does it come from?

https://github.com/adafruit/Adafruit_CircuitPython_SSD1306/blob/main/adafruit_ssd1306.py#L116

```python
# self.poweron
0xAE | 0x01

# self.init_display
0xAE,  # off
# address setting
0x20,
0x10  # Page Addressing Mode
if self.page_addressing
else 0x00,  # Horizontal Addressing Mode
# resolution and layout
0x40,
0xA0 | 0x01,  # column addr 127 mapped to SEG0
0xA8,
self.height - 1,
0xC0 | 0x08,  # scan from COM[N] to COM0
0xD3,
0x00,
0xDA,
0x02 if self.width > 2 * self.height else 0x12,
# timing and driving scheme
0xD5,
0x80,
0xD9,
0x22 if self.external_vcc else 0xF1,
0xDB,
0x30,  # 0.83*Vcc  # n.b. specs for ssd1306 64x32 oled screens imply this should be 0x40
# display
0x81,
0xFF,  # maximum
0xA4,  # output follows RAM contents
0xA6,  # not inverted
0xAD,
0x30,  # enable internal IREF during display on
# charge pump
0x8D,
0x10 if self.external_vcc else 0x14,
0xAE | 0x01,  # display on
```

https://github.com/adafruit/micropython-adafruit-ssd1306/blob/master/ssd1306.py#L40

```python
# self.init_display
0xAE | 0x00,  # off
# address setting
0x20, 0x00,  # horizontal
# resolution and layout
0x40 | 0x00,
0xA0 | 0x01,  # column addr 127 mapped to SEG0
0xA8, self.height - 1,
0xC0 | 0x08,  # scan from COM[N] to COM0
0xD3, 0x00,
0xDA, 0x02 if self.height == 32 else 0x12,
# timing and driving scheme
0xD5, 0x80,
0xD9, 0x22 if self.external_vcc else 0xF1,
0xDB, 0x30,  # 0.83*Vcc
# display
0x81, 0xFF,  # maximum
0xA4,  # output follows RAM contents
0xA6,  # not inverted
# charge pump
0x8D, 0x10 if self.external_vcc else 0x14,
0xAE | 0x01  # on
```

https://github.com/adafruit/Adafruit_CircuitPython_DisplayIO_SSD1306/blob/main/adafruit_displayio_ssd1306.py#L43

```
_INIT_SEQUENCE = (
    b"\xAE\x00"  # DISPLAY_OFF
    b"\x20\x01\x00"  # Set memory addressing to horizontal mode.
    b"\x81\x01\xcf"  # set contrast control
    b"\xA1\x00"  # Column 127 is segment 0
    b"\xA6\x00"  # Normal display
    b"\xc8\x00"  # Normal display
    b"\xA8\x01\x3f"  # Mux ratio is 1/64
    b"\xd5\x01\x80"  # Set divide ratio
    b"\xd9\x01\xf1"  # Set pre-charge period
    b"\xda\x01\x12"  # Set com configuration
    b"\xdb\x01\x40"  # Set vcom configuration
    b"\x8d\x01\x14"  # Enable charge pump
    b"\xAF\x00"  # DISPLAY_ON
)
```

TimonPeng, 2022-09-07:
Theoretically the ssd1306 and ssd1309 are partially interchangeable, but the different drivers above cause different results, I would like to know what the standard is..

markbirss, 2023-03-01:
@TimonPeng 

I found and used this other driver with my SDD1309 Display
https://github.com/smittytone/SSD1306OLED-Python