adafruit/Adafruit_CircuitPython_SSD1306 #77: Adafruit_SSD1306 library not working on micropython with blinka

Andrei631, 2022-10-07:
I am trying to use a SSD1306 display with a raspberry pi pico w, and i can't get it to work because of this error:
[image]
```
Line 70: super().__init__(buffer, width, height)
...
Line 327:  super().__init__(
Line 328:             memoryview(self.buffer),
Line 329:             width,
Line 330:             height,
Line 331:             external_vcc=external_vcc,
Line 332:             reset=reset,
Line 333:             page_addressing=self.page_addressing,
                 )
```

tekktrik, 2022-10-07:
Looks like the `format` argument isn't given to MicroPython's `framebuf.FrameBuffer`, whereas it's auto-populated in CircuitPython.  I can submit a fix for this, would you be willing to test it for me?

Andrei631, 2022-10-08:
> Looks like the `format` argument isn't given to MicroPython's `framebuf.FrameBuffer`, whereas it's auto-populated in CircuitPython.  I can submit a fix for this, would you be willing to test it for me?

Yeah, I am willing to test it.

Andrei631, 2022-10-08:
I edited the `super().__init__(buffer, width, height)` to `super().__init__(buffer, width, height, framebuf.MVLSB)` and now it's working, thanks for the help!

PR #78 (2022-10-09): Fix compatibility with MicroPython
Fixes #77 which is caused by the fact that the format must be specified for MicroPython, but not CircuitPython.

Links to docs for reference:
[MicroPython `framebuf.FrameBuffer`](https://docs.micropython.org/en/latest/library/framebuf.html#framebuf.FrameBuffer)
[CircuitPython `adafruit_framebuf.FrameBuffer`](https://docs.circuitpython.org/projects/framebuf/en/latest/api.html#adafruit_framebuf.FrameBuffer)
closed 2022-10-09 by tekktrik