adafruit/Adafruit_CircuitPython_SSD1306 #83: "Incompatible .mpy file" error while using 8.x library version on CircuitPython 8.2.7

cgobat, 2023-10-21:
Attempting to `import adafruit_ssd1306` when using the .mpy file from `adafruit-circuitpython-ssd1306-8.x-mpy-2.12.15` with CircuitPython 8.2.7 yields the following error:
```pytb
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_ssd1306.py", line 21, in <module>
ValueError: Incompatible .mpy file. Please update all .mpy files. See http://adafru.it/mpy-update for more info.
```

tannewt, 2023-10-23:
What board? Are you sure it is 8.2? 9.x will produce this error (use the .py file for now w/int.)

cgobat, 2023-10-23:
Yep, I am sure it is 8.2 (at least, 8.2.7 is what it displays when I enter the REPL). The board is a [Seeed Studio XIAO-RP2040](https://www.seeedstudio.com/XIAO-RP2040-v1-0-p-5026.html).

RetiredWizard, 2023-11-04:
Line 21 from adafruit_ssd1306 is attempting to import a Micropython library named framebuf.mpy. There is a Circuitpython frame buffer library called adafruit_framebuf.mpy which will be loaded if framebuf.mpy isn't found in either the root folder or the /lib/ folder. It would seem that you have a file called framebuf.mpy on your flash (either root or /lib/) that has an incompatible mpy format.

Which ever version of the framebuf library you attempt to use it must be compiled with a compatible 8.x mpy-cross. I'd recommend you don't have duplicate versions in either the root folder or the /lib/ folder and double check that the version of either framebuf.mpy or adafruit_framebuf.mpy is 8.x (you may actually have to cross compile the Micropython version manually if that's the one you intend to use, I really don't know what's involved in using the Micropython library in Circuitpython).

RetiredWizard, 2023-11-04:
Actually, it looks to me like the will be solved by installing /lib/adafruit_famebuf.mpy and removing any copies of framebuf.mpy 

I suspect the import of framebuf.mpy was intended to allow this library to be used in Micropython, not for the Micropython version of framebuf.mpy to be used in Circuitpython.

dhalbert, 2023-11-04:
`framebuf` is a native C module in MicroPython (from `extmod/modframebuf.c`), so there won't be a `.mpy` file. @cgobat: do you have any file named  `framebuf.mpy` on your CIRCUITPY? If so, where did it come from?

cgobat, 2023-11-04:
No, there is no `framebuf.mpy` on my CIRCUITPY drive.

dhalbert, 2023-11-04:
@cgobat I saw a reply from you in my email, but I think you deleted it. Did something change in what you said?

I installed the latest `adafruit_ssd1306.mpy` on a board running CircuitPython 8.2.7, and could not reproduce the original problem.

cgobat, 2023-11-04:
Yeah, I replied but then I realized my setup has changed since I opened this issue so I need to do some digging and roll it back to figure out what all I had on my CIRCUITPY at the time.

cgobat, 2023-12-01:
I've been unable reproduce the setup that was causing this error. I will reopen if it crops up again.

closed 2023-12-01 by cgobat